jneilliii/OctoPrint-PlotlyTempGraph #1: No Grpahs & No Custom Values

Dak0r, 2020-11-26:
Hi,
So I'm on 1.4.2 and seeing this merged PR I assume I don't need any custom version anymore?
https://github.com/OctoPrint/OctoPrint/pull/3506

I installed the plugin via Plugin Manager using the GitHub Url. I saw the new Graph-Box on the Temps-Tab, but it was empty (neither Tool nor Bed was shown.)

jneilliii, 2020-11-26:
Was your printer connected? That's usually the only reason that nothing is displayed in the graph. You will still have to create a separate single file plugin to add the data to the graph. There are a couple of examples here in the repo demonstrating that.

Dak0r, 2020-11-27:
Thanks for your fast response! I double-checked this, just to be sure. 
The printer is connected and I can see the current values below the graph, but the graph is empty. I have an Anycubic i3 mega with Marlin 1.1.19
[Here's a screenshot](https://imgur.com/a/HiCjlvN)

Are there any dependencies that I have to install or such?
I only installed the zip via ocotopis plugin manager.

jneilliii, 2020-11-27:
Thanks, there's definitely something weird going on here for me as well. I just tested in 1.5.0rc3 and am getting strange results from the data being returned by OctoPrint to my graphing library and I'm getting a chamber line when I don't have a chamber. If you could check something for me that would be helpful. If you open the developer console of your browser (typically F12 on your keyboard) and look at the console tab while refreshing the page are you getting any errors?

Dak0r, 2020-11-28:
I had a print running yesterday (with your graphs plugin disabled) and re-enabled it today to check for any errors.
Suddenly everything seems to be working fine, the graphs are displayed. No Errors in the console. I can not also open the legend (the button did nothing before) and see the values. I also see an "actual chamber" entry there.

Besides starting the print, I did not change anything between the last unsuccessful attempt and today.
I did run apt-get update and apt-get upgrade via ssh in an attempt to get it working, but it still didn't work after restarting the system. I then disabled the plugin and started the print.
Today I re-enabled it and it was working.

I tried to break by disabling and enabling it again, but it's still working.
I also tried to break it by doing a full reboot, but it's still working.

Next I installed and tried your "test plotly graph.py" example successfully.

I don't now has solved the issue, maybe it was indeed the update & upgrade in combination with a full shutdown and boot of the system. I'm pretty sure I rebooted the pi via ssh after upgrading the packages, but maybe I had to  needed a reboot and

jneilliii, 2020-11-28:
Yeah, I noticed the chamber thing yesterday and not sure where that's coming from. I've made progress on removing the memory leak though and have a develop branch with those changes. Now if you leave the browser open it will cut the graph correctly based on the time cut off in settings and not grow indefinitely.

Dak0r, 2020-11-28:
Sounds good, I have the version from the dev branch now running for about an hour. No initial issues, Will keep it installed for now 👍 

I just stumbled upon a different thing and opened a new issue for that to keep the issues clean

Dak0r, 2020-11-30:
Hi,
I was eventually able to reproduce the "No visible graphs" issue (does not seem to be a 100% reliable, though).

1. Everything is running fine, graphs are visible
2. I click the Disconnect Button in Octoprints Ui
3. I click the Connect Button
4. I refresh the page (F5).
5. All graphs are gone.

This also allowed me to take a look into Dev Console and actually found an error message! :)
I attached the log file 👍 
[console-export-2020-11-30_11-51-6.txt](https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/files/5615494/console-export-2020-11-30_11-51-6.txt)

jneilliii, 2021-01-16:
I think I have this issue fixed with the above commit, in the 0.0.4rc5 release. If you switch to the Development release channel in OctoPrint's Software Update settings it should give you an update notice.

closed 2021-01-24 by jneilliii

Dak0r, 2021-01-24:
Sorry for not responding sooner. I installed 0.0.4rc5 and for now it's working great, without any issues, thanks!

jneilliii, 2021-02-07:
Hey @Dak0r I know how to fix your dilemma with adding new features to the settings for the sensors so you don't have to remove them first. If you look at my TPLinkPlugin, I've had to do this a couple of times over the lifespan of the plugin. The specifics are found in the code [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/dd3568665a182452c456a694a5ba2f00dc542d19/octoprint_tplinksmartplug/__init__.py#L228-L295). You should be able to see where I'm iterating over a list of settings, adding a new default value to each dictionary and resetting it during the migration of version. I think for it to work you do have to increment both the version in setup.py and this `get_settings_version` callback.

jneilliii, 2021-02-07:
and actually looking at the enclosure plugin, it has had to do the same thing as well, so you just need to expand what's already there and increment some versions.

https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs/blob/31cc544144bb5aea882c3b58a50829cf16bafdf3/octoprint_enclosure/__init__.py#L123-L145

Dak0r, 2021-02-09:
Thanks a lot!! 
I'm currently a little busy - but I'm planing to implement and test it this week, maybe even today :)