jneilliii/OctoPrint-PlotlyTempGraph #2: New Temperatures-Keys only shown after Octoprint Restart

Dak0r, 2020-11-28:
I forked the Enclosure Plugin and implemented the "octoprint.comm.protocol.temperatures.received" hook here:
https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs

In general, it seems to be working fine for now!

Only If I make changes to my Enclosure configuration (adding a sensor or changing names of existing sensors) these changes won't show up in the Graph until I restart Octoprint.

I assume there's an issue if 'parsed_temps'-keys change at runtime, as both actions result in a new key being added to the dict when the hook is called for the next time.

Sorry if this is actually an issue with Octoprint? 🤷‍♂️

jneilliii, 2020-11-28:
can you try it with the same way I did it in the examples using parsed_temps.update?

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/8036e1c1574c2181b8d50806fad956680cbd0106/test%20plotly%20graph.py#L6

Dak0r, 2020-11-30:
You mean using the update method?
I'm afraid this does not work because I get a "keywords can't be an expression" error if I try to use a variable name as the key.

I can try to implement it, like in your fan-speed example where you put the values in a different dict first and then use update:
https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/8036e1c1574c2181b8d50806fad956680cbd0106/test_plotly_graph_fan_speed.py#L20

I will report back with my findings

jneilliii, 2020-12-02:
Yeah, let me know how it goes with that method.

Dak0r, 2020-12-03:
Hi, I wrote somewhat of a minimal working example for you to test with.
It adds a new Graph every 10 seconds and uses the update call. 

The result is the same as with my enclosure fork: the new graph shows up in the legend for a second but then gets removed on the next call. It's never displayed as a graph.

```python
# coding=utf-8
from __future__ import absolute_import

import octoprint.plugin
import random
import time

class TestPlotlyGraphDelayedCountChange(octoprint.plugin.StartupPlugin):
    def __init__(self):
        self.graph_values = dict()
        self.add_graph_time = 0 # force execution on first call 
        self.graphIndex = 0

    def temp_callback(self, comm, parsed_temps):
        if time.time() > self.add_graph_time:    # adds a new graph to graph_values dict, every few seconds
            self.graphIndex += 1
            self.graph_values["Graph"+str(self.graphIndex)] = (random.uniform(50,100),(random.uniform(0,50)))
            if self.graphIndex < 50:              # just add 50, please restart Octoprint to test further. More Graphs slow down the website
                now = time.time()
                self.add_graph_time = now + 10 # repeat 10 seconds in the future

        parsed_temps.update(self.graph_values) # update parsed temps using update
        return parsed_temps

__plugin_name__ = "Test Plotly Graph Delayed Count Change"
__plugin_pythoncompat__ = ">=2.7,<4"
__plugin_version__ = "0.1.0"
__plugin_implementation__ = TestPlotlyGraphDelayedCountChange()
__plugin_hooks__ = {
	"octoprint.comm.protocol.temperatures.received": __plugin_implementation__.temp_callback
}
´´´´

jneilliii, 2020-12-12:
@Dak0r I'm curious if you were to send the data more than a single time if it would graph. I think it might be getting detected as a single point and I don't believe those show up if there's not an additional point to graph.

jneilliii, 2020-12-13:
Have you noticed with your changes the following error in the developer console on initial page load? 

```
Error getting cutOffCount: TypeError: Cannot read property 'x' of undefined
```

It seems a refresh of the page squares it away but I don't remember that being an issue before. It was related to the plugin limiting the amount of time being graphed based on the cutoff value configured in the temperature section of OctoPrint.

Dak0r, 2020-12-13:
Hi, yes. I experienced an issue before I implemented my changes and it prevented the graphs from being displayed properly.
(I had an Exception in the Console log that I could trace back to this line) -> https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/077be1f8cf5d801ea3fd7617bb41d4953aa984c1/octoprint_plotlytempgraph/static/js/plotlytempgraph.js#L299

Then instead of fixing the cause of the exception, I just surrounded the affected line with a try-catch and wrote a console.error message if it happens. Defaulting the variable to 0 in such situations. This fixe made the graph work properly for me, even though it's not an ideal solution.

> @Dak0r I'm curious if you were to send the data more than a single time if it would graph. I think it might be getting detected as a single point and I don't believe those show up if there's not an additional point to graph.

Actually the values keep getting set on every call, as they don't get removed from the 'graph_values' dict, they just keep the same value and do not change.
With my changes, I can also see all "past"-data points after I refresh the page, it's just that they don't show up without reloading.

Thinking of it, maybe this is somehow related to the cutoff error? Maybe the new values get filtered instantly somehow...

jneilliii, 2020-12-13:
> Thinking of it, maybe this is somehow related to the cutoff error? Maybe the new values get filtered instantly for some reason...

Yeah, this is probably the underlying cause I think. That cutoff value is there in order to restrict the amount of time the graph is showing, so it doesn't just keep going until you run out of memory like before. That was added in a recent update hoping to get this ready for release [here](https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/commit/3a9ac339eefd09ce17477f6dbc6d41b7124ca70c). This cutoff isn't happening when you refresh and it pulls the data in the self.fromHistoryData callback.

jneilliii, 2020-12-13:
I did just find a way to reliably reproduce this issue. On initial page load when there is no printer connection the graph is initialized and there are no lines. Then once you connect the printer you get your caught error, and then the document.getElementById('plotlytempgraph').data starts getting populated. Refreshing the page then loads the data from history on page refresh.

jneilliii, 2021-01-16:
I think I have this issue fixed with the above commit, in the 0.0.4rc5 release. If you switch to the Development release channel in OctoPrint's Software Update settings it should give you an update notice.

closed 2021-01-24 by jneilliii