jneilliii/OctoPrint-PlotlyTempGraph #3: Legend showing chamber temperature atthough no chamber is present.

LMS0815, 2020-12-02:
Thank you for this wonderful plugin. Working well on my system (1.5.0)

I found a minor issue.
My printer has no chamber (properly reconfigured). The graph is not generated, which is the expected behavior.
But the legend of the graph is provided. I would expect not seeing it.

[image]

jneilliii, 2020-12-02:
Yeah, I'm seeing the same thing, as discussed in #1. I haven't yet had the time to dig into why that is happening, but suspect there is something different in the core OctoPrint values I'm reacting to that has that in there initially. Was meaning to ask @foosel about this.

LMS0815, 2020-12-02:
I assume you only need to look at the printer profile:

[image]

You just need to evaluate the chamber setting.
Same with the nozzle/tool setting.

jneilliii, 2020-12-03:
yeah, that would be a patch on my end for the issue but bypasses the point of this plugin, being able to graph data provided by any other plugin (like the fork of enclosure plugin that's being worked on), and it just graphs what it receives from octoprint.

LMS0815, 2020-12-03:
OK, got your point. It's more cosmetics than an issue anyway.

Never the less, the original OctoPrint graph is not showing the chamber in the legend.
Nor is it showing/plotting a second tool (See other reported issue https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/issues/4).

jneilliii, 2020-12-12:
Thanks to @Dak0r and the PR above this should be resolved in the latest 0.0.4rc4 release. If you can try the latest rc4 version available by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking install an report back that would be helpful.

```
https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/archive/0.0.4rc4.zip
```

jneilliii, 2021-01-24:
I have confirmation from a forum member that the combination of your fork and my recent updates that this is working now.

closed 2021-01-24 by jneilliii