jneilliii/OctoPrint-PlotlyTempGraph #4: Unneeded Tool graph for shared hotend

LMS0815, 2020-12-02:
I have a printer with two extrudes but a shared hot end (tool). 
There is no need to provide/generate a second graph for a non existing tool.

[image]

[image]

jneilliii, 2020-12-12:
I've linked you to the rc4 version in the other two issues you opened. Could you let me know if those changes also fixed this issue or not?

jneilliii, 2021-01-16:
@LMS0815 curious with your shared nozzle, does the temperature report the same for both tools all the time or is the second tool not graphed at all? It's a little hard to tell from the screenshot if they might be overlapped or not.

LMS0815, 2021-01-17:
Both share the same temperature (more or less as there seems to be a small deviation). I have no idea where the data is coming from, as my Printer (Creality CR-X)  is only reporting one nozzle temperature.

jneilliii, 2021-01-17:
Ok, thanks. I'll try to debug this using the virtual printer with a 2 nozzle shared setup and see how it acts.

jneilliii, 2021-08-28:
@LMS0815 finally got around to working on this plugin actively again now that it's released and have resolved the issue with additional nozzles being graphed with the shared nozzle option enabled. Change the plugin's release channel in OctoPrint's Software Update settings to Release Candidate and upgrade to version 0.1.2rc2 when prompted. if you want to try it out and verify.

PR #15 (2021-09-08): 0.1.1
* fixes an issue where the custom background defined in Custom Background plugin does not show, resolves #11 
* don't graph additional tool temperatures if reported by firmware when using shared nozzle, resolves #4
* add the ability to customize the name and color of lines in graph, resolves #12
closed 2021-09-08 by jneilliii