jneilliii/OctoPrint-PlotlyTempGraph #5: Minimum temperature from Settings not used.

LMS0815, 2020-12-02:
Seems the graph is not using the minimum graph temperature provided in OctoPrint configutation page, which is (in my case) 300°C.
[image]

[image]

jneilliii, 2020-12-02:
It will auto-grow to the size the graph needs to display the data. I will look into seeing if setting that based on the settings will effect the auto-growth or not, and if not implement that change.

jneilliii, 2020-12-12:
After looking at this in the English translation I think you misunderstand that setting in OctoPrint. I understand that to be the length of time in minutes to show on the graph, not the minimum temperature to graph.

jneilliii, 2021-01-24:
So plotly doesn't have a good way to do a dynamic range with setting a minimum value for an axis and allowing it to grow above that value. There is an open feature request, but it's requesting sponsorship ($10k-15k) so pretty sure I'm not covering that...lol. If I find a hack around this later down the road I will implement. In the meantime I have set the ability to configure the max temp to graph on the y axis and any temps above that configured limit get cut capped at the top of the graph. If the value is set to 0, the graph will resize automatically. Ready for the next release after a little more testing. If you want to try it out, you can copy/paste the URL below into Plugin Manager > Get More > ...from URL and clicking Install to overwrite your current installation.

```
https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/archive/0.1.0rc6.zip
```

jneilliii, 2021-01-24:
I'm not happy with this new option as it causes the left and right plot borders to be removed, and appears to be an upstream bug with Plotly. I've opened an issue on their github [here](https://github.com/plotly/plotly.js/issues/5442) for reference. Until that gets fixed upstream this will have to live that way when setting the max height of the graph if you so choose to enable that option.

jneilliii, 2021-01-29:
Fix available with version 0.1.0rc7.

closed 2021-01-29 by jneilliii