jneilliii/OctoPrint-PlotlyTempGraph #6: Please consider adjusting horizontal width to the available space.

LMS0815, 2020-12-07:
The plotting area is not using the available horizontal width. It seems that it is limited to the original OctoPrint layout.
If the width is bigger (e.g. extended with [UI Customizer](https://github.com/LazeMSS/OctoPrint-UICustomizer)) the horizontal width is not adjusted.

[image]

jneilliii, 2020-12-07:
So this one is actually a quick flag change, implemented in the above commit and will be included in the next release. If you want to test it out and verify you can copy/paste the URL below into plugin manager > get more > ...from URL and clicking Install. That will upgrade your exiting version to 0.0.4rc2.

```
https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/archive/develop.zip
```

LMS0815, 2020-12-08:
Thank you. I tested the version you provided.
[image]

 No difference. (even after cache refresh of the browser).
[image]

jneilliii, 2020-12-08:
Should have worked. I'll have to install that other plugin and experiment.

LMS0815, 2020-12-08:
Thank you. 
PS: If I compare the images, there is a small different in size, but its just not using all available space.

jneilliii, 2020-12-12:
@LazeMSS I'm wondering if you might be able to help me with this one. I made some updates to the plotly config to be responsive and changed the container to use min-width instead of setting the width and on initial page load in my develop branch but the graph is still being restricted on initial page load. If I resize the window or minimize and restore the window the graph will resize appropriately. Any pointers you can think of I can look at?

jneilliii, 2020-12-12:
I think I may have figured out a workaround for this. If you can try the latest rc4 version available by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking install an report back that would be helpful. I determined that this is related to a timing issue where UI Customizer is reacting to onAllBound and I'm creating the graph onAfterStartup, which happens prior to UI Customizer's changes. I added a Plotyly relayout command to the onAfterTabChange callback which will happen after the containers are resized. 

```
https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/archive/0.0.4rc4.zip
```

LazeMSS, 2020-12-13:
@jneilliii  does it not trigger that when I call the resize on all the tabs ?  $('#tabs').trigger('resize');

jneilliii, 2020-12-13:
I'm not sure why it's not resizing properly now with my last updates. It is here for me on initial page load with only the left sidebar and a full width right side with 0 in the left sidebar, at least in Chrome. 

[image]

[image]

Resizing the page also causes the graph to regenerate and scale to the page as expected. 

[image]

LMS0815, 2020-12-14:
rc4 seems to be OK for me.

[image]

jneilliii, 2020-12-14:
Great, the changes will be included in the next official stable release. Need to try and figure out a couple of other little quirks before release though unrelated to this issue.

closed 2021-01-24 by jneilliii