jneilliii/OctoPrint-PlotlyTempGraph #9: Cutoff value stopped working in 0.0.4rc5

Dak0r, 2021-01-27:
Hi, I'm currently using the 0.0.4rc5 release.

I noticed that the cutoff stopped working for me, if you leave Octoprint open, in this release:

I open Ocotoprints website, start a print and let it do its work. When I come back to the website later, I can see the whole graph (see image).

When I reload the page, the cutoff is correct for all existing values, but once new values come in, old ones do not get removed from the graph.

[image]

jneilliii, 2021-01-27:
Thanks I was noticing the same. I'm not quite sure why that's happening yet and haven't had time to dig in deeper.

jneilliii, 2021-01-29:
Fix available with version 0.1.0rc7.

closed 2021-01-29 by jneilliii