jneilliii/OctoPrint-PlotlyTempGraph #11: [FR]

fnsign, 2021-08-23:
Any chance to make it work with the Custom Background plugin?

jneilliii, 2021-08-23:
I'll try.

jneilliii, 2021-08-27:
Just released version 0.1.1rc1 that should in theory allow the custom background plugin to work. The only caveat is that you have no control of the same settings and it would be best to resize the image accordingly ahead of time. Change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted. if you want to try it out and verify.

fnsign, 2021-08-28:
Thanks for the feedback and the new version!
I currently have a 10h print going on, so I'm looking forward testing it tomorrow.

fnsign, 2021-08-28:
My current print broke, so I had the opportunity to test the new version.
It works! It took over my current custom background image without any changes.
Thanks again!

PR #15 (2021-09-08): 0.1.1
* fixes an issue where the custom background defined in Custom Background plugin does not show, resolves #11 
* don't graph additional tool temperatures if reported by firmware when using shared nozzle, resolves #4
* add the ability to customize the name and color of lines in graph, resolves #12
closed 2021-09-05 by jneilliii