jneilliii/OctoPrint-PlotlyTempGraph #12: Is there a way to customize the tool name?

paul-tian, 2021-08-24:
Hello jneilliii,

Is there a way to customize the tool name? Currently it shows "tool 0" etc. in the graph and would lead to a bit confusing.
It would be great if the "tool 0" could be customize to the name of presenting data source (e.g. "extruder" for "tool 0")

jneilliii, 2021-08-24:
Right now I'm just using it as it's given to me from octoprint's side I think. It shouldn't be too difficult to have a name mapping overlay for the graph though.

jneilliii, 2021-09-05:
Just released version 0.1.1rc3 that should in theory allow the customization of the label and color of the graph lines. Change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted. if you want to try it out and verify.

PR #15 (2021-09-08): 0.1.1
* fixes an issue where the custom background defined in Custom Background plugin does not show, resolves #11 
* don't graph additional tool temperatures if reported by firmware when using shared nozzle, resolves #4
* add the ability to customize the name and color of lines in graph, resolves #12
closed 2021-09-08 by jneilliii