jneilliii/OctoPrint-PlotlyTempGraph #14: Is it possible to add a logging function?

kdkwarhead, 2021-08-31:
A Start/stop logging button that ask for a logfile name. It then logs time, target and actual temperatures in a csv file. This is useful to compare different cooling solutions or different PID settings.

I use the serial log for this but you need to remove the old serial file and restart the Pi in order to get a new serial log file. I then use the following SerLogCleaner.bat file to clean up the serial.log file and its messy:

rem Instruction
rem Move SerLogCleaner.bat and serial.log in to empty directory (not a network disk)
rem Run SerLogCleaner.bat
rem Load serial_temp_only.log into excel and use space as separator
rem kdkwarhead 2020-12-08

rem Copy temperatures lines to temp1.txt
findstr /c:"Recv:  T:" serial.log > temp1.txt

rem Copy only lines that are not set to 0C to temp2.txt
findstr /v /r /c:"/0.00 ....... /0.00" temp1.txt > temp2.txt

rem Fix the file for Excel (change "-replace ' ',';'" to "-replace ' ',','" if your spredsheet program uses , as sepaerator)

powershell -Command "(gc temp2.txt) -replace 't:','' -replace 'b:','' -replace '/','' -replace ',','.' -replace ' ',';' | Out-File -encoding ASCII serial_temp_only.csv"

rem Delete temp files
del temp1.txt
del temp2.txt

jneilliii, 2021-08-31:
That sounds like a plugin of its own and out of scope for the intent of this plugin. I hope you understand.

closed 2021-08-31 by jneilliii

kdkwarhead, 2021-08-31:
Ok. I just thought that would be easy to add for someone like you that already done the difficult things.
Thanks anyway for this plugin-

jneilliii, 2021-08-31:
creating the plugin itself wouldn't be that difficult to do, I just think it's better off as a separate one. It could actually be done in a one-file plugin I think. If I get some time later today I can see if I can come up with something for you. The biggest concern is writing file contents while printing, as that can in some cases cause issues and is one of the reasons why serial logging is not enabled by default and only recommended while debugging. If you don't see an issue on your end with that, then a plugin would work.

kdkwarhead, 2021-08-31:
I only turn on serial logging when I do this kind of tests. I know that serial logging can cause problems but it has worked fine for me.
Logging temperatures is not a super important thing, especially since nobody else has requested it. Don’t put too much work in to in on my behalf. I just thought that it would be easily to implement since you plugin already collects the data.

I can use the serial log and my batch file to get the curves in to excel for my own test. I’m not skilled enough to do my own plugin and I can’t really ask you to make plugin that only I use a few times when I change somethings in my printer.

jneilliii, 2021-08-31:
Here's the framework of what you are looking for. Would just need some small adjustments to the write function to put it somewhere else. 

https://gist.github.com/jneilliii/cef912c3df93724b5da17595729ad681

kdkwarhead, 2021-08-31:
Thanks,
I know some programing languages (like C, C++ and Pascal) but I never tried Python. Python seems to be fairly easy to read but I’m sure it takes quite some time to learn to really program something useful with it for me. I’m a machinery engineer so programing is something I do only for fun (very small projects).

Thanks for al the information and your effort but I don’t think that I will ever finish a plugin for octoprint.