jneilliii/OctoPrint-PlotlyTempGraph #16: Save state of legend toggle.

technoplunk, 2021-09-16:
Thanks for this new graph plugin I'm loving it already. It would be nice if one of the configuration options would be a toggle to "Always show legend". I prefer it on as I have a few extra sensors in my printer and its confusing without it. Everytime I reload the octoprint interface I have to turn it back on.

jneilliii, 2021-09-16:
Nice idea. I will add that with the next release when I get around to it.

jneilliii, 2021-09-18:
Rather than make it an option, it makes sense to me to just make the visibility state persistent. I've done that in the above commit and available as part of version 0.1.2rc1. You can switch the release channel for the plugin to `Release Candidate` in OctoPrint's software update settings.

technoplunk, 2021-09-18:
This also works for me thanks!

PR #18 (2021-09-18): 0.1.2
* make legend visibility persistent, resolves #16
* add optional hidden field to name map, resolves #17
closed 2021-09-18 by jneilliii