jneilliii/OctoPrint-PlotlyTempGraph #17: Hide Specific Graph Lines/Legends

technoplunk, 2021-09-16:
I have some extra temp sensors in my printer that don't have corresponding heaters. (chamber, edge of plate). Octoprint doesn't really have a way to configure them so they don't also have corresponding graph lines/legends for their "target" temperature, I only need the "actual" lines. 

As such would it be possible to have "hide" checkboxes in the "Name Mapping" section of this plugin's config that would then hide these extra lines/legends that would otherwise never have data?

jneilliii, 2021-09-16:
I will have to consider this one for a bit to figure out the best way to handle this case.

jneilliii, 2021-09-18:
I've added the hidden option for the name map in the above commit and available as part of version 0.1.2rc2. You can switch the release channel for the plugin to `Release Candidate` in OctoPrint's software update settings.

technoplunk, 2021-09-18:
Works good thanks!

closed 2021-09-18 by jneilliii

JLatasa, 2021-09-18:
Hi. After upgrading from 0.1.1 to 0.1.2 I can not see anymore the "plotly menu" that usually shows on the top right-hand side of the graph.

jneilliii, 2021-09-19:
That hasn't changed since this plugin was made, the modebar has always been disabled.

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blame/44b90cc6b391da79866ebe4e2a3bc2365cded9f2/octoprint_plotlytempgraph/static/js/plotlytempgraph.js#L51

PR #18 (2021-09-18): 0.1.2
* make legend visibility persistent, resolves #16
* add optional hidden field to name map, resolves #17