jneilliii/OctoPrint-PlotlyTempGraph #20: Second tool not showen

EddyMI3d, 2021-09-19:
Hello @jneilliii !

With the latest update (V 0.1.2), the second tool does not appear in the graph any more.

[image]

jneilliii, 2021-09-19:
Can you try adding the second tool to the name map?

aerickson, 2021-09-21:
I thought that this was a bug also. 

It's hard to figure out the additional temperature values (I was missing 'actual'... in my plugin I don't add this). I think the previous behavior of displaying all is better/less confusing (with users adding a config entry if hiding a value is desired).

Is there a way to inspect the temperature names in the javascript console?

Thanks for all of your work. :)

jneilliii, 2021-09-21:
I agree, this is less than ideal but number of users with multiple extruders is probably not that high. I will plan on figuring out a way to automatically pick up what is returned from the printer and add to the name map though.

EddyMI3d, 2021-09-21:
> 
> 
> Can you try adding the second tool to the name map?

Yes, that worked for me too. Thanks you

jneilliii, 2021-10-05:
Adding note, that this also effects the Enclosure Plugin fork as sensors aren't automatically added to the graph anymore with the name map. Will be digging into this further over the weekend to remedy.