jneilliii/OctoPrint-PlotlyTempGraph #21: How to add a target temperature

littlezaphod, 2021-09-24:
Hi!
First of all congrats to that awesome plugin!
I got all my temp sensors showing in the plot:
[image]
However I am struggling to also get the chamber target temperature to show. I am missing something with the name mapping, can you give me a clue?
[image]
[image]
cheers Pat

jneilliii, 2021-09-24:
Are you using the enclosure plugin or another approach for getting the temperature data? My assumption is that the enclosure plugin doesn't report the target temp, so therefore isn't available in the data I'm reacting to for generating the graph possibly.

littlezaphod, 2021-09-24:
I am using the enclosure plugin fork: Enclosure Plugin with Support for Custom Graphs

jneilliii, 2021-09-24:
Yeah, it doesn't report target temps. 

https://github.com/Dak0r/OctoPrint-Enclosure-with-Custom-Graphs/blob/75ac72cf9b26f2e585b7d7ca036daf0f6983663d/octoprint_enclosure/__init__.py#L1961