jneilliii/OctoPrint-PlotlyTempGraph #22: Compatibility problem with TouchUI

sirinsidiator, 2021-09-27:
I noticed that TouchUI and PlotlyTempGraph do not play together. I already reported the issue to TouchUI, but I'm not sure if it wouldn't be easier to solve it on this plugins side. For example by adding a div block with the id "temp" so its expectation is met: https://github.com/BillyBlaze/OctoPrint-TouchUI/issues/515

jneilliii, 2021-09-27:
Since I am using the replaces method of the templating system in OctoPrint, I could definitely attempt this adjustment. I'll give it a try and see if it works as expected by wrapping the tab's contents in another div.

jneilliii, 2021-09-27:
Unfortunately, it doesn't appear to be that easy of a fix. Just adding the wrapper div still doesn't play nice with TouchUI.

jneilliii, 2021-09-27:
This will have to be fixed on the TouchUI side I'm afraid.

closed 2021-09-27 by jneilliii

sirinsidiator, 2021-09-28:
Thanks for the quick response and checking into it. I've also noticed another "issue", but that's more of a feature request.
The current zoom behaviour on the graph doesn't work the way I'd expect from a touch interface. Something like swipe to pan and pinch to zoom would be more appropriate.
I'm pretty sure this is outside the scope of this plugin though, so I'll see if I can put something together on my end and make a pull request once it works.

jneilliii, 2021-09-28:
Yeah, all the functionality is built into the plotly js library, so if you find a way to make it more touch friendly, feel free to submit a PR.

jneilliii, 2021-10-24:
After further testing, this version isn't going to work either, don't waste your time. This is something that BillyBlaze will have to fix in TouchUI.