jneilliii/OctoPrint-PlotlyTempGraph #24: Graphs not present in Name Map are Hidden by default

Dak0r, 2021-10-13:
Hi, long time no see! :)

I just updated the plugin on my setup and noticed my Enclosue Plugin Graphs were gone, until I added them to the new name map via the settings menu. 

From a user perspective I think it would be better if graphs not listed in the map would be visible by default and can be hidden using the settings? This way you can also copy & paste the name from the temp view to the name map.

I think the issue is, that `self.lookup_visibility` always returns false for graphs not present in the name_map.
I haven't tested the fix, yet, but I think it just a small change in this line:

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/44b90cc6b391da79866ebe4e2a3bc2365cded9f2/octoprint_plotlytempgraph/static/js/plotlytempgraph.js#L101

to  `if (!name_map || (name_map && !name_map.hidden())) {  `

And again, thank your for this plugin!

PS: I noticed in your issues section, that people seem to actually use my enclosure fork! So I decided that I should finally take the time to work on that PR...

jneilliii, 2021-10-13:
Hey @Dak0r, duplicate of #20

closed 2021-10-13 by jneilliii

jneilliii, 2021-10-16:
So @Dak0r I think your change may actually break to addTrace of Plotly for some reason. I noticed that temp lines load initially but new received temps aren't getting added. I'll have to dig into this later, unless you potentially see a reason why that may be the case.

Dak0r, 2021-10-17:
Argh, too bad. I just had another look at the source code and could not find any obvious issue related to this.

I feel like this would only happen if the name_map_-values don't match with the index that is used to when extending the trace. So for some reason a wrong index is used, maybe because one of the other look-up functions does not work as we expect them?

Though the only thing I noticed (I don't know if this is related though), the names in the name_map must be unique, otherwise the line https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/44b90cc6b391da79866ebe4e2a3bc2365cded9f2/octoprint_plotlytempgraph/static/js/plotlytempgraph.js#L381 could return a wrong index and adding or extending traces would fail.

So it could make sense to check in line 79, if the label in the map is empty, to sort out "empty" entries:

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/44b90cc6b391da79866ebe4e2a3bc2365cded9f2/octoprint_plotlytempgraph/static/js/plotlytempgraph.js#L79

for example ```if(name_map && name_map.label()  !== '') {```

It could be worth a try, but also feels a little far-fetched as a solution for this issue :/

jneilliii, 2021-10-17:
Yeah, that doesn't seem to make a difference. Looks like I'll have to go the other route as I was planning anyway to just add the additional name map on unindexed temp data.