jneilliii/OctoPrint-PlotlyTempGraph #28: Enhancement - Add additional Firmware Temps to graph

jalanjarosz, 2021-11-14:
This is a request to enhance the OctoPrint graph adding additional Firmware Temps.

Klipper FW allows the user to configure additional temperature sensors: https://www.klipper3d.org/Config_Reference.html?h=pins#temperature_sensor

The configured sensors can be called anything and would be displayed upon calling a M105.

My printer config is as follows:
```
# Temperature from the host machine
[temperature_sensor cpu]
sensor_type: temperature_host
min_temp:0
max_temp:100
gcode_id:C
#sensor_path:
#   The path to temperature system file. The default is
#   "/sys/class/thermal/thermal_zone0/temp" which is the temperature
#   system file on the Raspberry Pi computer.
```

I've currently named my RaspPi's M105 ID as "C" so it would be graphed as the "Chamber Temp".
```
Send: M105
Recv: ok B:18.7 /0.0 C:51.0 /0.0 T0:19.1 /0.0
```

I propose allowing the plug-in to config custom M105 id's like "cpu" or "mcu" and graphing them.  Under the plug-in options, under the "Name Mapping" section, add a column for custom M105 id.

Example M105 output with a custom id:
```
Send: M105
Recv: ok B:18.6 /0.0 T0:19.0 /0.0 rpi:50.0 /0.0
```

I'd be willing to test any changes as well
Thank you.

jneilliii, 2021-11-14:
That's the beauty of this plugin, it allows you to create a single file plugin that will do this for you. If I have time, I'll try to throw one together for this specific example, in the meantime if you have python programming knowledge you could use the existing examples as a guide and do it yourself. The one linked below would be a perfect starting point as it is basically reacting to the recv commands from the printer for a specific pattern. 

https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/master/test_plotly_graph_fan_speed.py

jneilliii, 2021-11-17:
I've put together an example single file plugin that with my virtual printer testing seems to work. To install copy/paste the URL below into Plugin Manager > Get More > ...from URL and click install. Once you restart the graph should start to show those temps. 
```
https://raw.githubusercontent.com/jneilliii/OctoPrint-PlotlyTempGraph/master/klipper_additional_temp.py
```
[image]

jalanjarosz, 2022-01-27:
Sorry for the delay on getting back to you on this.

I downloaded the klipper_additional_temp.py example you provided and tested it with the MCU temp in a new board I'm working with.  The temp's did not show up.      

Note: the MCU/shield combination I'm using basically has 4 stepper drivers on it and some free pins.  No other sensors.

After a couple OctoPrint refreshes, I finally got the temp of the MCU reported in the graph although the graph was not updating.

Next, I added the SBC CPU temp sensor to the same .py, removed and installed the updated version and nothing came up the graph.

After some tinkering, I discovered that the OctoPrint temp graph does not parse/update unless there is a TOOL or BED temp being reported.  I'm not sure if this is intended or not?

Anyway, I went back and decided to split my temp sensors into individual .py files so they can be added/removed when/if necessary.

In this graph, the "BED" is the actual SBC CPU Temp.
Metro M4 MCU is the temp of my Adafruit Metro M4 board.  

[image]

I think your example will work and I'm okay with closing the ticket now, or I can report back after I move the Metro M4 to my Production printer, which then I'll put multiple temp's in the graph.

jneilliii, 2022-01-27:
The plugin depends on octoprint's temperature polling for sure. Thanks for reporting back.

closed 2022-01-27 by jneilliii