jneilliii/OctoPrint-PlotlyTempGraph #29: Chamber temp not appearing

anrienexe, 2022-05-28:
Everything was working fine until I updated Octopi yesterday. I don't often update so I don't know which version I was but I am now on version 1.8.1.

Bed temp should be green:

[image]

Only tool + bed graph are shown :( :

[image]

jneilliii, 2022-05-28:
Need to see how the temperature is coming in from the terminal tab of octoprint

anrienexe, 2022-05-29:
Is this what you are looking for?

```
#######################################
#### Enclosure Temp Sensor Section ####
#######################################
[thermistor chamber_temp]
temperature1: 25
resistance1: 10000
beta: 3950

[temperature_sensor enclosure_temp]
sensor_type: chamber_temp
sensor_pin: P0.23
min_temp: 0
max_temp: 100
gcode_id: C
```

jneilliii, 2022-05-29:
Not much experience with klipper, but what I'm looking for is this bits from the Terminal tab from OctoPrint. That is how OP will give the information to the plugin to allow for processing properly. 

[image]

jneilliii, 2022-05-29:
I was able to reproduce the issue with 1.8.1 and virtual printer emulating with chamber temperature. I'm not quite sure why the plugin isn't picking up the temp now from the serial comms but suspect it does have something to do with changes to core OctoPrint. Will investigate further. 

[image]

jneilliii, 2022-05-29:
I think I remember the issue now after looking through old OctoPrint issues. You have to also enable the chamber in the printer profile.

[image]

With that option enabled then the virtual printer gets the data from OctoPrint and is able to graph.

[image]

anrienexe, 2022-05-29:
Thanks for you speed light support! :)

closed 2022-05-29 by anrienexe