jneilliii/OctoPrint-PlotlyTempGraph #37: Enhancement: Additional Temperature filter to supplement of Y-axis limit

DTMRi, 2022-08-13:
An optional filter for "freak" temperature reports from the printer, i.e., to filter an obv. unrealistic “59000 °C" reading instead of capping it off with a limit on the axis and the resulting ugly vertical spikes in the graph (see pic for reference).

In my opinion this could be a convenient software fix for a cosmetic hardware problem, that has the potential of tedious and/or expensive debugging. Thanks for the cool plugin!

[image]

DTMRi, 2022-08-14:
Something like a "sanity check" in form of a max allowable derivative to filter unrealistic jumps of the actual temperature would be most elegant I think, at least when doing It in Matlab, where all my “coding” experience lies. 
You are far more qualified to judge about the actual implementation, if you even like the idea ^^

jneilliii, 2022-08-14:
Yeah, I think there's an actual example of that in the docs in the form of a single file plugin. Not sure how the ordering would work out though in the processing loop. You could copy/paste the URL below into Plugin Manger > Get More > ...from URL and clicking Install. Look like it will limit values to below 300. 
```
https://github.com/OctoPrint/Plugin-Examples/raw/master/sanitize_temperatures.py
```

DTMRi, 2022-08-14:
Oh cool, thanks!
This only solves half the problem, as it doesn’t filter the low spikes. Nonetheless you can close the issue, if you don’t think it is necessary to implement such a filter into the graph.

jneilliii, 2022-08-14:
Actually....

> The following example shows how to filter out actual temperatures that are outside a sane range of 1°C to 300°C.

DTMRi, 2022-08-14:
Then:

closed 2022-08-14 by DTMRi