jneilliii/OctoPrint-PlotlyTempGraph #38: [BUG] Random OOOOOOOOT etc instances keep adding

LariOesch, 2022-09-11:
Random lines keep addning from time to time, al of them start with random number of O's and ending with T

[image]

jneilliii, 2022-09-11:
This is a side effect of Prusa Firmware and their use of temperature reporting to transmit MMU status data. I have a companion plugin you can install to resolve that, where it basically filters those out completely. From plugin manager > get more, copy/paste the URL below into the ...fromURL field and click install. 

```
https://gist.github.com/jneilliii/6a2cbf0056e8c9e5ed5a828fa1694aad/raw/0bfa67bbc2f7cd543e3d83fce3cd60cdcfad1b00/prusa_temp_fix.py
```

Duplicate of pinned post #30

closed 2022-09-11 by jneilliii

LariOesch, 2022-09-11:
Ah cool, thanks. Sorry for duplicate.