jneilliii/OctoPrint-PlotlyTempGraph #40: [BUG] Graph Key overlaps right vertical axis numbering

puterboy, 2022-12-22:
Would it be possible to add space between the graph key and the right vertical axis numbering?

Even better would be to add these features:
1. Ability to select placement of the key (e.g., bottom vs. right hand side)
2. Ability to toggle on/off the key

BTW thanks for all the improvements -- looks great!!!!

jneilliii, 2022-12-22:
I think legend positioning is possible with plotly.js library. will investigate further. I may end up deciding doing something similar to what I'm doing with Bed Level Visualizer and providing an advanced section in settings that allows for overriding the data and layout parameters of plot generation, and that would also allow for tweaking for your other requests relative to font color.

puterboy, 2022-12-22:
Ability to turn off the legend would also be nice as I would rather have more room for the graphs :)
Also, it is often difficult to match colors to legends anyway.

Losing the fixed legend wouldn't be much of a loss, since when you hover, you basically get a pop-up legend ties the sensor name to each trace at the given time point -- which is much easier to align and mentally parse anyway. So basically the fixed legend is often redundant.

jneilliii, 2022-12-22:
> Ability to turn off the legend would also be nice as I would rather have more room for the graphs :)

that's what the button does in the bottom right-corner...

jneilliii, 2022-12-22:
just noticed that the reason you're getting the overlap is because of your mixed temp settings (using fahrenheit and celsius). Without that enabled it doesn't overlap the numbers because they don't exist...

[image]

puterboy, 2022-12-22:
> > Ability to turn off the legend would also be nice as I would rather have more room for the graphs :)
> 
> that's what the button does in the bottom right-corner...
Doh! My bad :)

puterboy, 2022-12-27:
There still seems to be at least a minor bug...
Whenever I refresh the Octoprint screen, the graph appears without the right hand side labels.
They only appear if I toggle the legend on-and-off.

Specifically, sequence is as follows:

- Initially, no RHS axis values (and no legend)
- Toggle legend on, then RHS axis values appear, overlapped by legend
- Toggle legend off, then RHS axis values appear cleanly (without legend as expected)

closed 2023-02-12 by jneilliii

puterboy, 2023-02-15:
Hi, I know you don't have more time to invest in "Fahrenheit", but personally, I much preferred the even if imperfect RHS scale than subplots. The whole purpose for me was to see all the plots on the same graph but to have a left and right-sided scale so that I can easily convert between and visualize both Fahrenheit and Celsius.

Even if you understandably don't wish to put in new effort, perhaps you could revert this change. I honestly don't see any logic for Fahrenheit vs. Celsius subplots.

Thanks for all the great plug-ins!

PR #43 (2023-02-12): 0.1.4
- TouchUI compatibility enhancements
- update plotly.js library to version 2.9.0
- add several single file plugin examples for adding additional temperature data to graph
- adjustments related to sizing of bundled custom background images, resolves #35
- adjustments related to coloring grid lines and labels based on custom background plugin settings, resolves #36
- changed hovermode to x, allowing any hover position to show trace pop-ups, resolves #42
- use sub-plots rather than right side axis, resolves #40
- added hover text font color option, resolves #41
- Fahrenheit conversion and d [... cut]