jneilliii/OctoPrint-PlotlyTempGraph #41: [FEATURE] Ability to set text color for the "boxed" current value of each graph

puterboy, 2022-12-22:
Currently, the font color for the data shown inside the colored box for each graph seems to always be the theme font text color (in my case blue). However, depending on the graph color, there may be very little contrast between the graph color (background of the text box) and the data value (font color of the text).

This make some numbers very hard to read.

2 potential suggestions:
1. Either allow text color to be specified for each graph (perhaps add a column parallel to the one used to set the graph line color)
2. Automagically (i.e., mathematically) choose the maximally contrasting color - e.g., one furthest away on color scale (if this makes sense????)

jneilliii, 2022-12-22:
I think I understand what you mean, but could you share a screenshot?

jneilliii, 2022-12-22:
Is this happening with the latest 0.1.4rc7 release? I know I've tweaked some things for text color stuff, and maybe I just tweaked it a little too far.

puterboy, 2022-12-22:
Yes I have that version (I am subscribed to the release channel).
BTW, I have noted this problem from before your tweaking -- as Plotly gives a broad color range of colors and it is very difficult to have one font color that contrasts well with all of them -- unless you use only dark colors and say a white font or vice-versa

jneilliii, 2022-12-22:
cool, just double-checking as I thought you might be. I'll have to investigate the options on this one too with plotly. I've never really looked at any of there graphs with a large data set. I think generally they set the font color to white for those.

closed 2023-02-12 by jneilliii

puterboy, 2023-02-15:
Not sure if there is something I am doing wrong, but changing the text color ('txt') for individual graphs doesn't seem to have any effect -- i.e. they all stay the same. (note that changing the tag color ('clr') is working fine for me.)

I don't believe I have any conflicting themeify settings btw.

jneilliii, 2023-02-15:
try refreshing the page after making the change.

puterboy, 2023-02-15:
I refreshed multiple times to no avail.

jneilliii wrote at about 22:01:23 -0800 on Tuesday, February 14, 2023:
 > try refreshing the page after making the change.
 > 
 > -- 
 > https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/issues/41#issuecomment-1430797730
 >

jneilliii, 2023-02-15:
That does confuse me a little bit then. I just double-checked and set my bed actual to use red and got this after page refresh.

[image]

PR #43 (2023-02-12): 0.1.4
- TouchUI compatibility enhancements
- update plotly.js library to version 2.9.0
- add several single file plugin examples for adding additional temperature data to graph
- adjustments related to sizing of bundled custom background images, resolves #35
- adjustments related to coloring grid lines and labels based on custom background plugin settings, resolves #36
- changed hovermode to x, allowing any hover position to show trace pop-ups, resolves #42
- use sub-plots rather than right side axis, resolves #40
- added hover text font color option, resolves #41
- Fahrenheit conversion and d [... cut]