jneilliii/OctoPrint-PlotlyTempGraph #42: Display latest temperature when mouse over right edge of graph

jmsaltzman, 2023-02-10:
It would be great to watch the values update in real time, but it's almost impossible to position the mouse left-right to show the latest time. Then when mousing over the area just to the right (but still inside the border), the labels go away :(  Thank you!

jneilliii, 2023-02-10:
not sure what you mean, but if I hold the mouse over the end of the graph like in the below picture, it automatically updates for me as the temp changes and the timeline moves to the left. Also, you do know the current real-time temp is displayed just below it in the Actual column?

[image]

jneilliii, 2023-02-10:
I just realized what you mean, once the temp starts changing the closest line moves away from the mouse and therefore doesn't show the hover pop-up anymore. It seems I can fix this by applying a hovermode = x to the plotly library and anywhere you hover on the graph will show all the pop-ups at once. I'm going to implement this now and test, then push to rc branch. If you change the plugin's release channel in OctoPrint's Software Update settings to `release candidate` you will be notified when that update goes live.

jmsaltzman, 2023-02-11:
Yes! Thank you! Mousing near the right black border to show all the curves, each with a little colored tag tracking the current value is just.. [chef kiss].

PR #43 (2023-02-12): 0.1.4
- TouchUI compatibility enhancements
- update plotly.js library to version 2.9.0
- add several single file plugin examples for adding additional temperature data to graph
- adjustments related to sizing of bundled custom background images, resolves #35
- adjustments related to coloring grid lines and labels based on custom background plugin settings, resolves #36
- changed hovermode to x, allowing any hover position to show trace pop-ups, resolves #42
- use sub-plots rather than right side axis, resolves #40
- added hover text font color option, resolves #41
- Fahrenheit conversion and d [... cut]
closed 2023-02-12 by jneilliii