jneilliii/OctoPrint-PlotlyTempGraph #46: Eliminate or Replace degree symbol and C/F suffix?

ScottWell1, 2023-08-09:
Not sure if this is an enhancement request, or if I am just missing something obvious. The example on the project's home page shows numbers without a degree symbol or C/F suffix, but I cannot find a way in settings to suppress these.  

I ask because...  I would like to display some non-temperature values (specifically an enclosure fan speed to start with) returned as 0-100% from a custom single-file plugin.  So minimally, I would like to suppress the degree symbol and C/F suffix for those (as shown on the project's home page), though preferably for just specific sensors rather than all of them.  Better yet, it would be nice to define a custom suffix (like "%") for specific sensors.

Is there any way to do that?

jneilliii, 2023-08-10:
Yeah, when changes were made to incorporate Fahrenheit vs Celsius it added the symbols. Prior to that, which is what the screenshots are from, didn't do that. Will look into possible options for setting the suffix per "temperature" line instead of basing it on degree type.

jneilliii, 2023-08-10:
I also meant to mention that you may want to check out the Top Temp plugin. I think you can define custom scripts to run as well for making small graphs.

ScottWell1, 2023-08-10:
> Yeah, when changes were made to incorporate Fahrenheit vs Celsius it added the symbols. Prior to that, which is what the screenshots are from, didn't do that. Will look into possible options for setting the suffix per "temperature" line instead of basing it on degree type.

Got it - I thought that might be the case (old screenshot before features added).  Thank you for looking into it.  Not sure many other users will want to add non-temp values, so I will understand if you determine it is not worth the effort. :-)

ScottWell1, 2023-08-10:
> 

> I also meant to mention that you may want to check out the Top Temp plugin. I think you can define custom scripts to run as well for making small graphs.

I am using top temp, and it works great (calling same bash scripts), but of course the charts only show each sensor in isolation.  Then I discovered PlotlyTempGraph and realized how useful it would be to have them on the same graph, so i can easily relate changes in one sensor (like enclosure temp) to another sensor (like controller board temp) or how each is affected by fan speed.  Nice having it all in one spot!

jneilliii, 2023-08-11:
I've been able to achieve this change, but need to think about how to integrate this in the settings. There's not really any room left for adding a text box. If you want to try it out you can switch the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and update when prompted. Once upgraded you can manually modify config.yaml with the command `nano ~/.octoprint/config.yaml`, find the `hover_template` property in the name_map settings you want to modify and adjust it like below. Press ctrl+s to save and then ctrl+x to exit. Then restart OctoPrint. 

```
  plotlytempgraph:
    _config_version: 5
    max_graph_height: '0'
    max_graph_height_sub: '300'
    name_map:
    - color: ''
      convert_to_fahrenheit: false
      hidden: false
      hover_color: ''
      hover_template: '<b>%{y:.1f}%</b> '
      identifier: tool0 actual
      label: ''
      use_fahrenheit: false
      use_sub_plot: false
```
The above makes the hover template show percent sign on Tool 0 like this.

[image]

ScottWell1, 2023-08-11:
I changed release channel to Release Candidate, but checking for updates responded "Everything is up to date."  

I see the 0.1.5rc1 on your Github rc branch, but Octoprint Software Update still shows 0.1.4 for both Installed and Available.  Restarted Octoprint, verified I had selected Release Candidate, and tried both "check for updates" and "force check for updates" -- same result, "Everything is up to date."

Did I miss a step, or ?

jneilliii, 2023-08-11:
Sorry about that. I must have not changed the branch when creating the tag/release. I just fixed it and verified update check is seeing new version now. Use the force check for updates and you should get it now.

ScottWell1, 2023-08-11:
Thanks, I got the 0.1.5rc1 now.  Looks great!  

I did notice one other thing about the 0-values (issue #47), so I will add another comment on that issue.

Thank you!!!

[image]

closed 2023-08-11 by jneilliii