jneilliii/OctoPrint-PlotlyTempGraph #50: Not able to show values from mini plugins

thomas82kj, 2023-09-29:
I have been trying for some time now to use this plugin to graph some more values. But i have simply not been able to get anything more than the tool and bed temperature on the graph, which have been there the whole time.

I started by modifying the heater example plugin to fit repetier firmware (the order of @: and B@: is reversed). I believe my modified version should work now, i have been able to see that the values gets picked up in the log files: 
2023-09-27 18:40:50,877 - octoprint.plugins.heater_temps_rep - DEBUG - {'nozzle_heater': (36.0, None), 'bed_heater': (255.0, None)}

Despite this I get nothing on the graph. Today i also tried the test plugin which generate random values: https://github.com/jneilliii/OctoPrint-PlotlyTempGraph/blob/b5f5a4f4fad5c3da892d457ea25ae40654d5f1e8/test%20plotly%20graph.py

To my surprise this does not add any values to the trend either.
Any clue where to look next?
I am using plugin version 0.1.5 on Octopi 0.18.0 (Octoprint 1.9.2)

jneilliii, 2023-09-29:
Should work that way. Might be easier for me to see your code and get you to share an example temp report line from the terminal tab to debug further. It's odd the random example doesn't work either. You don't have any errors in octoprint.log?

thomas82kj, 2023-09-29:
There is actually quite a few errors in the console (I have copied the beginning of some of them, from firefox)

Error plotting history data for bed_heater [packed_plugins.js:9652:25](http://192.168.87.118/static/webassets/packed_plugins.js?e87aab25)
    fromHistoryData http://192.168.87.118/static/webassets/packed_plugins.js?e87aab25:9652
    callViewModelIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1798
    callViewModelsIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1792
    Pn http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:30
    ur http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:51
    callViewModelsIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1792
    callViewModels http://192.168.87.118/static/webassets/packed_core.js?06860971:1791
    _onHistoryData http://192.168.87.118/static/webassets/packed_core.js?06860971:1672
    _ifInitialized http://192.168.87.118/static/webassets/packed_core.js?06860971:1686
    _onHistoryData http://192.168.87.118/static/webassets/packed_core.js?06860971:1672
    propagateMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:24
    Pn http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:30
    ur http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:51
    propagateMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:24
    onMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:31

Error plotting history data for nozzle_heater [packed_plugins.js:9652:25](http://192.168.87.118/static/webassets/packed_plugins.js?e87aab25)
    fromHistoryData http://192.168.87.118/static/webassets/packed_plugins.js?e87aab25:9652
    callViewModelIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1798
    callViewModelsIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1792
    Pn http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:30
    ur http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:51
    callViewModelsIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1792
    callViewModels http://192.168.87.118/static/webassets/packed_core.js?06860971:1791
    _onHistoryData http://192.168.87.118/static/webassets/packed_core.js?06860971:1672
    _ifInitialized http://192.168.87.118/static/webassets/packed_core.js?06860971:1686
    _onHistoryData http://192.168.87.118/static/webassets/packed_core.js?06860971:1672
    propagateMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:24
    Pn http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:30
    ur http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:51
    propagateMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:24
    onMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:31

TypeError: a is undefined
    Ua http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:129
    p http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:134
    ec http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:134
    G http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    x http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:133
    Qd http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:134
    yd http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    zd http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    ha http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    Hc http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    notifySubscribers http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    xa http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    b http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    lookup_visibility http://192.168.87.118/static/webassets/packed_plugins.js?e87aab25:9350
    fromHistoryData http://192.168.87.118/static/webassets/packed_plugins.js?e87aab25:9611
    callViewModelIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1798
    callViewModelsIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1792
    Pn http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:30
    ur http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:51
    callViewModelsIf http://192.168.87.118/static/webassets/packed_core.js?06860971:1792

Error: You cannot apply bindings multiple times to the same element.
    p http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:131
    k http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:131
    vc http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:132
    c http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:133
    b http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:133
    c http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:133
    r http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:134
    ec http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:134
    G http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
    x http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:133
    Qd http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:134
[... 3 lines cut ...]
    Hc http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:130
[... cut ...]
    _onHistoryData http://192.168.87.118/static/webassets/packed_core.js?06860971:1672
    _ifInitialized http://192.168.87.118/static/webassets/packed_core.js?06860971:1686
    _onHistoryData http://192.168.87.118/static/webassets/packed_core.js?06860971:1672
    propagateMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:24
    Pn http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:30
    ur http://192.168.87.118/static/webassets/packed_libs.js?6bee1dee:51
    propagateMessage http://192.168.87.118/static/webassets/packed_client.js?3339ed72:24

jneilliii, 2023-09-30:
Yep, missed the hover_template in part of the code where the entry gets automatically added to the settings. Since it was missing caused the data to not load right or save the new entry into settings. Pushing update now and will release version 0.1.7 with fix.

thomas82kj, 2023-09-30:
I have just updated, and it seems to work like it should now. Thanks a lot for the very quick reaction.
Now my next issue will be to somehow filter the values, especially since my bed heater does on/off control, so it's really something like the average value over a minute or so that's interesting. I guess I will need to create some static variables in the mini plugin, and probably convert some strings to numbers to do an averaging or low pass algorithm.

closed 2023-09-30 by jneilliii