jneilliii/OctoPrint-PlotlyTempGraph #53: Duplicate plot entries

aaronevers, 2024-07-05:
I occasionally get duplicate plot entries. 

[image]

jneilliii, 2024-07-05:
Not 100% sure what could cause this to happen. Looks like you're running Prusa firmware based on what you have listed. If you look in settings do those additional duplicates also show to be able to set their properties or hide?

aaronevers, 2024-07-05:
Octoprint: 1.20.2; Plotly 0.1.7

Prusa MK3s, firmware: 3.13.3. Happens with both a MMU instance of Octoprint and a separate instance without it. Seems to happen at the same time to both. I haven't been able to determine if it's something in the terminal causing it.

I see no duplicate entries in the Name Mapping table.

jneilliii, 2024-07-05:
That's very weird. I'd be interested in seeing if there are any errors in the browser's developer tools > console tab when this happens. To determine if it's something related to the serial communication you could enable serial logging and then we can check out serial.log to see if there's anything that stands out where it may be reporting these duplicates. I have a sneaky suspicion that this may be a hard one to track down though due to some random race condition.

aaronevers, 2024-07-06:
[image]
[serial.log.zip](https://github.com/user-attachments/files/16114908/serial.log.zip)

Left the log going until I saw duplicates.  Printer was idle.

jneilliii, 2025-04-27:
Yeah, nothing in the serial.log looks out of place. Could be related to that websocket reconnection. It's doing plotTraces, and then when reconnects merged in fromHistoryData and then continues with plotTraces. The fromHistoryData should be all the previous plotTraces, but there could be some weird merging issue. Could be an upstream plotly bug, and it's been a while since I've updated the library.